ros/rosdistro #33039: OpenVDB Debian Buster

SteveMacenski, 2022-05-03:
Hi,

Apparently I've been receiving build failures on STVL for a number of weeks now that have been going to spam and I didn't see. I'm looking to resolve https://build.ros.org/job/Nbin_db_dB64__spatio_temporal_voxel_layer__debian_buster_amd64__binary/142/console failures and noticing something odd and I figure I'm missing something obvious that someone can point out.

These are debian buster builds and the error is a `KeyError: 'libopenvdb3.2'` that it can't find the `3.2` version of openVDB. Understandable, Debian Buster has `v5.2` https://packages.debian.org/buster/libopenvdb5.2 in its package manager. That leads me to believe I simply need to update the OpenVDB rosdep keys for Buster, but I noticed that it already has `libopenvdb5.2` as its key (https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L4274). So I'm a little confused by the Buster jobs are picking up `v3.2` which should only be done in Stretch or some very old versions of Ubuntu. I checked on the `-dev` version, they're also set to `v5.2` in Buster. 

Here's my package.xml https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/1.4.5/package.xml#L26-L27

Any ideas / thoughts?

mjcarroll, 2024-01-19:
After conversation, we determined that releasing OpenVDB as a vendor package is the best approach.  This is being tracked: https://github.com/SteveMacenski/spatio_temporal_voxel_layer/pull/267

nachovizzo, 2024-01-23:
Update: Now is being tracked here:

https://github.com/SteveMacenski/spatio_temporal_voxel_layer/pull/281