ros/rosdistro #33596: Discussion: python3 typing-extensions package

SubaruArai, 2022-06-16:
Hello, I'd like to add python3's official [typing-extensions](https://pypi.org/project/typing-extensions/) package which basically backports some typing functionalities.

The problem is, the system package for Ubuntu 20.04 is quite old, and is basically useless for what is does.

Fundamentally it is a problem caused of rosdep not handling dependency versions (or locks), but we can't change that in the short term.

I personally would like to make this an exception and use pip instead of system packages, I'm pretty sure that's against the rules.

Can a maintainer share his/her opinions?

jacobperron, 2022-07-14:
While we strongly prefer to use system packages, I think it is still possible to add a pip version of the package as long as the rosdep key follows the [naming convention](https://github.com/ros/rosdistro/blob/master/CONTRIBUTING.md#pip-1) (using a `python3-` prefix and a `-pip` suffix). The main downside to using pip packages is that any ROS packages that depend on them **cannot** be released on the buildfarm (which doesn't support pip dependencies).

In the case of adding a rule for typing-extensions, IMO we should add a rule for installing the available system packages. If this is not sufficient for you, then I think it would be okay to add a second rule for installing the pip version of the package. I.e. we would have two sets of rules, `python3-typing-extensions` and `python3-typing-extensions-pip`. However, other maintainers may disagree.

SubaruArai, 2022-07-14:
@jacobperron Thanks for the feedback! That does seems a logical compromise.

Not being able to use in the buildfarm might be a dealbreaker for some people, but then again we already have quite a lot of pip only packages, so...

This package is basically a backport of newer python version's `typing` package - which is it's fundemental problem.
i.e. a system installing `python3.8` will only have a backport up until `3.8`.

I would like to hear opinions from other maintainers, too.

tfoote, 2022-08-26:
Yeah, our standard policy is that if a system version exists we use that one. It's sometimes frustrating. But you have to remember that as we put these distributions together it's a large coordinated effort to make sure that everything works well together. And everyone wants their specific part to be upgraded, but everything else to be stable. And unfortunately that doesn't work if there's more than one person. For the releases we have to keep things stable. 

There's also separate major issues of pip installations overriding and breaking system installations. So that if you have a correct dependency and someone else pip installs a newer version it could break anyone with the correctly declared system dependency. 

Now if this is really critical to your workflow, you can always create your own set of rosdep rules that are used on your systems and you can know that you've QA'd these changes and they won't break your users. But we can't accept that for the general community.

closed 2022-08-26 by tfoote

SubaruArai, 2022-08-28:
@tfoote Thanks for the detailed explanation!

While it's a bummer for me, I totally understand that it's more important to maintain overall stability.
And yeah, the problem with pip installations breaking system ones, too.

> Now if this is really critical to your workflow, you can always create your own set of rosdep rules that are used on your systems and you can know that you've QA'd these changes and they won't break your users. But we can't accept that for the general community.
Thanks for providing an alternative! We might go down this route if really needed.

Overall, I can't thank you enough for the detailed (and time-consuming) reply!