ros/rosdistro #34385: How to patch ROS2 release.

Levi-Armstrong, 2022-09-08:
I went through the patch process for third party release but the following command fails `git-bloom-release foxy`. Is there another process that should be used?

Levi-Armstrong, 2023-03-15:
I was following the release process for a third party found [here](http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty)

Levi-Armstrong, 2023-03-15:
I am trying to release [this](https://github.com/tesseract-robotics-release/qt_advanced_docking_system-release) for foxy where I was able to create a patch for noetic but for some reason I was not able to for foxy.

nuclearsandwich, 2023-03-15:
This type of issue is probably best served by [ROS Answers](https://answers.ros.org) at the moment.

> following command fails `git-bloom-release foxy`

Can you please share the output of the failed command, as well as the values you entered when prompted for the upstream version and tag? For example, I had no problem re-releasing 3.8.2 from a forked version of the release repository.

nuclearsandwich, 2023-04-04:
@Levi-Armstrong any updates here?

nuclearsandwich, 2023-04-14:
Closing due to lack of response. Please re-open this issue or open a new one if you're still having problems.

closed 2023-04-14 by nuclearsandwich