ros/rosdistro #35490: Unable to establish SSL connection-raw.githubusercontent.com/ros/rosdistro/master/ros.key

anilbommareddy, 2022-12-01:
~$ wget https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -O - | sudo apt-key add -
--2022-12-01 14:37:25--  https://raw.githubusercontent.com/ros/rosdistro/master/ros.key
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
Unable to establish SSL connection.
gpg: no valid OpenPGP data found.

cottsay, 2022-12-07:
Is the date in your system clock correct?

build-error, 2022-12-14:
[image]
i am also facing a similar issue.

cottsay, 2022-12-20:
This is a connectivity problem between your system and GitHub, folks. Unless the commands are wrong in the documentation, I'm not really sure what I can do except walk you through some troubleshooting steps to figure out what's wrong with your systems.

Both work on my machine:
```
~$ wget https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -O - | sudo apt-key add -
--2022-12-20 04:29:38--  https://raw.githubusercontent.com/ros/rosdistro/master/ros.key
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133, 185.199.108.133, 185.199.109.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1167 (1.1K) [application/octet-stream]
Saving to: 'STDOUT'

-                              100%[==================================================>]   1.14K  --.-KB/s    in 0s      

2022-12-20 04:29:38 (81.0 MB/s) - written to stdout [1167/1167]

Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
OK
```

```
~$ curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.key | sudo apt-key add -
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
OK
```

Can you get to GitHub at all?
```
~$ curl https://raw.githubusercontent.com/ros/rosdistro/master/README.md 
This repo maintains the lists of repositories defining ROS distributions.

It is the implementation of [REP 143](http://ros.org/reps/rep-0143.html).

It also the home of the rosdep rules.

Guide to Contributing
---------------------

Please see [CONTRIBUTING.md](CONTRIBUTING.md).

Review guidelines
-----------------

Please see the [review guidelines](REVIEW_GUIDELINES.md) to look at the criteria to get a pull request merged into this repository.
```

Do you have the `ca-certificates` package installed so that you can establish SSL connections? Can you download the file using a browser instead? What version of Linux are you using?

tfoote, 2023-01-11:
CLosing this as it doesn't appear to be an issue with the distro. THis is a pretty generic SSL connectivity issue. Please search on answers.ros.org or even generic Stack Overflow for more information and if you can't resolve it searching there ask your own question on one of those sites.

closed 2023-01-11 by tfoote