ros/rosdistro #35653: Missing rosdep key for ament_cmake_clang_tidy

RFRIEDM-Trimble, 2022-12-14:
Hello.

I had a package relying on `ament_cmake_clang_tidy` but that key is not present in rosdistro. Looking around to see which packages use it, the only recent example was [here](https://github.com/sloretz/clang_tidy_examples/blob/8a3a2a2e870139e9390606b977cbd35224ff7c4b/ament_cmake/package.xml). 

The error:
`Cannot locate rosdep definition for [ament_cmake_clang_tidy]`

ROS Version: Galactic 

Is there a reason this is not present as a rosdep key  in this repo or other reason it cannot be found?

mjbogusz, 2022-12-22:
If I understand the index correctly, the rosdep package would be `ament_lint`, as per https://github.com/ros/rosdistro/blob/master/galactic/distribution.yaml#L156
Note that it pulls in multiple other packages in tow, but I think it's a [possibly intentional] limitation of rosdep when dealing with multiple packages originating from the same repository.

sloretz, 2022-12-29:
`ament_cmake_clang_tidy` is a ROS package, not a rosdep key. It looks like it is present in ROS Galactic. https://github.com/ament/ament_lint/tree/galactic/ament_cmake_clang_tidy

The error message you posted looks like `rosdep` was unable to find the package `ament_cmake_clang_tidy`, and then looked it up as a rosdep key and was unable to find it that way too. There's not enough context for me to reproduce it. Mind providing a minimal reproducible example?

Does the issue persist after `rosdep update`?

closed 2023-01-03 by RFRIEDM-Trimble