ros/rosdistro #35709: Noetic rm_control regression - source job fails to build

sloretz, 2022-12-29:
Nrel_import-package is failing with:

```
00:00:01.288 Importing package: ros-noetic-rm-referee:0.1.16-1focal:focal:source
00:00:01.288 running command ['reprepro', '-v', '-b', '/var/repos/ubuntu/building', '-S', 'misc', 'include', 'focal', '/var/repos/ubuntu/building/queue/focal/Nsrc_uF__rm_referee__ubuntu_focal__source__7755/ros-noetic-rm-referee_0.1.16-1focal_source.changes']
00:00:01.302 Unexpected binary character \001 in /var/repos/ubuntu/building/queue/focal/Nsrc_uF__rm_referee__ubuntu_focal__source__7755/ros-noetic-rm-referee_0.1.16-1focal_source.changes
00:00:01.302 There have been errors!
```
Not sure why yet. There are some unusual characters in the changelog: https://github.com/rm-controls/rm_control/blob/master/rm_referee/CHANGELOG.rst

sloretz, 2022-12-29:
The `CHANGELOG.rst` file does contain the byte 0x01

```
git clone git@github.com:rm-controls/rm_control-release.git -b release/noetic/rm_referee/0.1.16-1
```

Seems that way in the source repo too: https://raw.githubusercontent.com/rm-controls/rm_control/09a2fac73c7e11beefbdf39c2ebb6769fc8fe97a/rm_referee/CHANGELOG.rst

sloretz, 2022-12-29:
I think this will be fixed by https://github.com/rm-controls/rm_control/pull/88

tfoote, 2023-01-11:
As this has been triaged and submitted upstream i am going to close this. @sloretz You can consider removing it from the distro but it will automatically disable itself after a number of failed retries and only retry daily.

closed 2023-01-11 by tfoote