ros/rosdistro #36088: Missing rosdep key for pymap3d-pip in ubuntu jammy

pariaspe, 2023-02-10:
Hi all, I'm new here, sorry if I'm not in the right place.

I'm getting a jenkins build [error](https://build.ros2.org/job/Hdev__aerostack2__ubuntu_jammy_amd64/lastBuild/console), not finding pymap3d-pip key. rosdep rule is defined [here](https://github.com/ros/rosdistro/blob/4b7fb961ad829ef1433ff9e1b2b45a615e52ff00/rosdep/python.yaml#L405) and it seems right. Also our own CI action finds it, any help?

Should I open a PR adding ubuntu jammy version? Should I add a new rule for `python3-pymap3d` apt [pkg](https://packages.ubuntu.com/jammy/python3-pymap3d)? Thanks!

SteveMacenski, 2023-02-10:
The name in your package.xml is wrong `pymap3d`. The key is actually `pymap3d-pip` as you can see as the root name of the rosdep entries. This is what we look for in the package.xml parsing. Updating that should fix your issue.

Separately, we do generally prefer apt packages over pip packages, so submitting updates to use the apt version would be a separate and useful contribution

pariaspe, 2023-02-10:
Hey, thanks for the help.

We actually are using the key as you mention in the [package.xml](https://github.com/aerostack2/aerostack2/blob/main/as2_python_api/package.xml#L19).

Alright, I'll open a PR in the next days, glad to contribute!

SteveMacenski, 2023-02-10:
Oh I see (sorry, I scanned quickly and that's the most common reason), that actually looks like the key resolved. It looks like that doesn't actually exist on your OS... which is odd because I can pip install that on my 22.04 machine. That is odd. Can't say I know the reasons for that, might be something pythony (which I don't use extensively). Its perhaps best to get the opinions of another maintainer here that is more familiar with these kinds of issues w.r.t. python

pariaspe, 2023-02-21:
Solved in #36164 , thanks!

closed 2023-02-21 by pariaspe