ros/rosdistro #36442: Add sparkfun-qwiic to rosdep?

RnoldR, 2023-03-12:
Is it possible to add sparkfun-qwiic to ubuntu-64bit 22.04 for raspberry pi? It has to be added with pip3.

I am a newbee to ros and busy trying to learn it. I want to use it for my robots which are equipped by a Raspberry Pi with ubuntu 22.04 64bit. In my specific case I want to use the [sparkfun-qwiic python package](https://learn.sparkfun.com/tutorials/sparkfun-qwiic-button-hookup-guide/python-package) that is already installed with pip but is not recognozed by rosdep. There appears no way to add an external/unknown package to rosdep other than via the list on the github/rosdep repository. 

- is that really true or did I overlook something?
- if I didn't overlook something is it possible to add an option to rosdep like --add-3rdparty-lib <filename>.yaml to add unknown libraries?

Or is there a better way to add dependencies?

mjcarroll, 2023-03-13:
If you would like to add a key to rosdep, please follow the contributing guide for rosdep keys here: https://github.com/ros/rosdistro/blob/master/CONTRIBUTING.md#rosdep-rules-contributions

Pure python packages (not in pip) are also available via this mechanism.

closed 2023-03-13 by mjcarroll

RnoldR, 2023-03-21:
Thank you for sharing this link!