ros/rosdistro #36889: Deleted for rosdep rules EOL - RPI

rafal-gorecki, 2023-04-17:
Hi, I notice that there was deleted ubuntu focal dependency i last commit by @tfoote. If it's intentional, is it necessary to edit `sources.list.d` now?
(changes in last commit, line 4508 in python.yaml: https://github.com/ros/rosdistro/commit/59517bc9c4241f16ede721aff4878f5ab1b748b4#diff-d8721f41777cc41754c7a34c95d1ef9c8f6fbce7575d129bcbcd32383cde667bL4508)

tfoote, 2023-04-17:
That looks like an over match in the regex cleanup. If you're using it please submit a PR to add it back.

```
python-rpi.gpio-pip:
  debian:
    pip:
      packages: [RPi.GPIO]
  fedora:
    pip:
      packages: [RPi.GPIO]
  ubuntu:
    pip:
      packages: [RPi.GPIO]
```

rafal-gorecki, 2023-04-17:
I'm not sure if i do it in recommended way, but there is a link to PR: #36891

closed 2023-04-18 by rafal-gorecki