ros/rosdistro #37310: `attrs` completely missing for Ubuntu Jammy

Interpause, 2023-05-22:
After https://github.com/ros/rosdistro/commit/b88e07b7a3206ab31d6420d5f47c7fecef6e6c7e, `attrs` cannot be installed for Ubuntu Jammy anymore using either `python-attrs` or `python-attrs-pip`.

nuclearsandwich, 2023-06-05:
Both of those rosdep keys implicitly refer to python2 libraries, which are not available or supported on Ubuntu Jammy. 

There is a python3-attr package in Ubuntu Jammy: https://packages.ubuntu.com/jammy/python3-attr which could be added to rosdep under a new key: `python3-attr` for use with distributions supporting Python 3 (all ROS 2 distributions and ROS Noetic).

clalancette, 2023-09-07:
Since I don't think there is any further work to be done here, I'm going to close this out.  Feel free to reopen if I'm mistaken.

closed 2023-09-07 by clalancette