ros/rosdistro #37822: Support component-level keys for boost

Ryanf55, 2023-06-29:
If my interpretation of the[ ROS style guide](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Code-Style-Language-Versions.html#boost) is correct, a ROS developer should try to reduce their usage of boost to the minimum. Luckily, boost provides a COMPONENT level find and install. This has the advantage of requiring less boost to be installed on the system, and less linked in.

For example:
```cmake
find_package(Boost COMPONENTS program_options REQUIRED)
...
target_link_libraries(myLibrary PUBLIC boost::program_options)
```

Ideally, in my package.xml, I could specify only needing the program options component of boost list so:
```xml
<depend>boost-program-options</depend>
```
Which would install `libboost-program-options-dev`.

If this method of installing dependencies is desirable and would be accepted into rosdep, please let me know.

clalancette, 2023-08-02:
So we already have these kinds of keys in rosdep: https://github.com/ros/rosdistro/blob/935070ebcbde0fc28814dbc83adc059ea59e59f1/rosdep/base.yaml#L2648-L2670 .  Did you mean something else?

Ryanf55, 2023-09-03:
Some of them are missing, I'll contribute the ones I need.

closed 2023-09-03 by Ryanf55