ros/rosdistro #38767: Support for Bookworm

rosterloh, 2023-10-16:
I'd like to work on adding support for Debian bookworm to rosdistro. Is there a guide or a good starting point for this work?

nuclearsandwich, 2023-11-05:
Hi @rosterloh can you clarify what kind of support you are looking to contribute to?

Debian Bookworm is a supported distribution for the ROS Infrastructure although no current ROS distributions support it. We do not change the target platforms of stable distributions but it is possible that we'll update the Debian target for ROS 2 Rolling Ridley when we move it to Ubuntu 24.04 as well.

If you wanted to help with that effort, the first thing to do would be to start looking at rosdep keys for Debian Bookworm. There are rosdep repo checks that run as part of the GitHub Actions in this repository (ros/rosdistro) but you can run them locally to run them across the entire database rather than just changed files. Adding bookworm to those checks by updating [this config file](https://github.com/ros/rosdistro/blob/master/test/rosdep_repo_check/config.yaml#L63-L64) would be a good start!

rosterloh, 2023-11-07:
That's very helpful thank you. To demonstrate what I'm looking to support I've run with the following config 
```yaml
---
package_sources:
  debian:
  - !deb_base_url http://deb.debian.org/debian main
  - !deb_base_url http://deb.debian.org/debian contrib
  - !deb_base_url http://deb.debian.org/debian non-free
  - !deb_base_url http://repos.ros.org/repos/ros_bootstrap main
package_dashboards:
- pattern: !regular_expression .*//deb.debian.org/debian/.*
  url: https://packages.debian.org/{os_code_name}/{binary_name}
supported_versions:
  debian:
  - bookworm
supported_arches:
  debian:
  - arm64
```
I will start working through the missing deps and submit a PR

clalancette, 2026-02-10:
I'm going to close this out as there isn't a well-defined goal here.  Please feel free to keep working towards adding Bookworm support.

closed 2026-02-10 by clalancette