ros/rosdistro #38921: released package with too-high version number, what to do?

berndpfrommer, 2023-11-09:
I accidentally released a package with a higher version number than I wanted.
The package in question is event_camera_msgs, the distro where I messed up is rolling. Since this was my first package released I didn't realize that I should have used separate minor version numbers to distinguish between the distros, like so:

rolling: 1.0.5
humble: 1.1.5
iron: 1.2.5

But now I already have a package out there under rolling with version 1.1.4, and I [just submitted a PR](https://github.com/ros/rosdistro/pull/38919) to change that to 1.0.5. Is there any way to fix this cleanly such that going forward I can use 1.0.x for rolling?
Thanks!

clalancette, 2023-12-28:
> But now I already have a package out there under rolling with version 1.1.4, and I [just submitted a PR](https://github.com/ros/rosdistro/pull/38919) to change that to 1.0.5. Is there any way to fix this cleanly such that going forward I can use 1.0.x for rolling?

In short; not really.  I see in the meantime you've downgraded Rolling to 1.0.x, which is OK, but it is weird that rolling is behind both humble and iron.  My suggestion is that you update rolling to 1.3.x so it is ahead of both of them.

berndpfrommer, 2023-12-28:
Thanks much, that's what I'll do!

closed 2023-12-28 by berndpfrommer