ros/rosdistro #39863: AeroVironment MacCready Works REP-144 Name Prefix of "avmw_`

Ryanf55, 2024-02-09:
I would like to "reserve" a [REP-144 complaint name prefix](https://ros.org/reps/rep-0144.html#naming-rules) for  [AeroVironment MacCready Works](https://www.avinc.com/maccready-works). We don't yet have any packages published, but have an interest to contribute open source on things that would benefit the ROS community that aren't tied to any of our intellectual property.

Can we reserve a name prefix in rosdistro while we work on our packages internally so it doesn't get scooped from under us? The effect of getting scooped results in an ABI breaking change in many internal repos in the CMake target names and namespaces for all the code, which would be unfortunate, time-consuming, and costly for my team to have to deal with.

I propose the following name prefix:
`avmw_` = AeroVironment MacCready Works

avmw is not yet taken: https://github.com/search?q=repo%3Aros%2Frosdistro%20avmw&type=code

If this name is accepted, how could we go about reserving the prefix?

clalancette, 2024-03-04:
First of all, thank you for checking in about the names.  This will ensure that the names already meet REP-144.  From that perspective, the `avmw_` prefix for the packages seems fine.

As far as reserving it, we have no policies or procedures for doing that, and I really don't want to start.  At least for now, the community is small enough that I don't think it is warranted.

What I can suggest is that as soon as you have the packages ready, you open up PRs that at least add in the `source` entries for these packages.  That will at least ensure that you start using the names as soon as they are ready.

clalancette, 2025-01-02:
Given that this was a question, I'm going to close this.  Please feel free to reopen if you need more information.

closed 2025-01-02 by clalancette