ros/rosdistro #40236: rosdep fails with rolling

roncapat, 2024-03-18:
When issuing `ROS_DISTRO=rolling rosdep install -r -s` rosdep fails to find many packages. The result is that my `rolling` CI fails with messages like:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
package_x: No definition of [geometry_msgs] for OS version [jammy]
package_x: No definition of [image_common] for OS version [jammy]
package_x: No definition of [image_transport] for OS version [jammy]
package_x: No definition of [yaml_cpp_vendor] for OS version [jammy]
package_x: No definition of [pcl_conversions] for OS version [jammy]
package_x: No definition of [pcl_ros] for OS version [jammy]
package_x: No definition of [point_cloud_transport] for OS version [jammy]
package_x: No definition of [rosidl_default_generators] for OS version [jammy]
package_x: No definition of [sensor_msgs] for OS version [jammy]
package_x: No definition of [tf2_geometry_msgs] for OS version [jammy]
Continuing to install resolvable dependencies...
```

I'm using up-to-date `rolling-desktop-full-jammy` and `rolling-desktop` images.

Am I right to suspect this has some kind of linking with the preparation of rolling to switch on top of `Ubuntu 24.04 Noble`?

rcp1-beg, 2024-03-25:
Same here, e.g., rosdep fails for actuator_msgs:
`rosdep resolve actuator_msgs --rosdistro=rolling
WARNING: given --rosdistro rolling but ROS_DISTRO is "humble". Ignoring environment.

ERROR: No definition of [actuator_msgs] for OS version [jammy]

No definition of [actuator_msgs] for OS version [jammy]
	rosdep key : actuator_msgs
	OS name    : ubuntu
	OS version : jammy
	Data:
_is_ros: true
		debian:
		  bookworm:
		    apt:
		      packages:
		      - ros-rolling-actuator-msgs
		osx:
		  homebrew:
		    packages:
		    - ros/rolling/actuator_msgs
		rhel:
		  '9':
		    dnf:
		      packages:
		      - ros-rolling-actuator-msgs
		ubuntu:
		  noble:
		    apt:
		      packages:
		      - ros-rolling-actuator-msgs
`

mjcarroll, 2024-04-12:
Yes, things are in flux with the transition to 24.04.  Once migration is finished, things should quiet down a bit.

mjcarroll, 2024-05-15:
@roncapat and @rcp1-beg can you check and see if things are "back to normal" here?

roncapat, 2024-05-15:
@mjcarroll I still have CIs based on top of `osrf/ros-rolling-desktop` failing because `hardware_interface` package is not found.

roncapat, 2024-05-15:
According to https://hub.docker.com/r/osrf/ros/tags?page=&page_size=&ordering=&name=rolling the rolling images have been updated 1 month ago, not sure if it's related to this.

rcp1-beg, 2024-05-15:
> @roncapat and @rcp1-beg can you check and see if things are "back to normal" here?

You mean by using rolling for jammy / 22.04 *or* for noble / 24.04?

clalancette, 2024-06-11:
Given that we have now switched rolling over to Noble completely, I'm going to close this out.