ros/rosdistro #40272: Geographiclib does not exist in older versions of Ubuntu

Louis-PR, 2024-03-20:
Since [#40244](https://github.com/ros/rosdistro/pull/40244), we have trouble issues using rosdep install with geographiclib dependency in a Bionic Beaver version of Ubuntu.

Refering to this [line](https://github.com/ros/rosdistro/blob/bede3a3127935c5e364322dc04f63a935e35a36d/rosdep/base.yaml#L959), @clalancette, could we add: 

    artful: [libgeographic-dev]
    bionic: [libgeographic-dev]
    vivid: [libgeographic-dev]
    wily: [libgeographic-dev]
    xenial: [libgeographic-dev]
    yakkety: [libgeographic-dev]
    zesty: [libgeographic-dev]

to the "geographiclib" key for ubuntu in base.yaml below : 

ubuntu:
    '*': [libgeographiclib-dev]

?

clalancette, 2024-03-20:
Bionic (and older) are all end-of-life now, so I don't think we should keep those keys around.  But I'll ask others what they think; @ros/rosdeputies do you think we should restore these keys for end-of-life distributions?

wico-silva, 2024-03-22:
I see this output in the terminal:

```
Looking for packages in '/opt/ros/melodic/share'... using matcher with tags [melodic, ubuntu, bionic]
loading cached data source:
	https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
	/root/.ros/rosdep/sources.cache/a4b3c8956ab7a3e6d293ba33b57fac42799e69e6.pickle
loading cached data source:
	https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
	/root/.ros/rosdep/sources.cache/b4931581ca7c5741f7ee4ac9767a1174838296f7.pickle
loading cached data source:
	https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
	/root/.ros/rosdep/sources.cache/7cfbeed211a3f9a6507c86cebaeedb3e3c4387f8.pickle
loading cached data source:
	https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
	/root/.ros/rosdep/sources.cache/825fae71ab9ea366e0b22e8dbe1e2948232cb2e8.pickle
loading cached data source:
	https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
	/root/.ros/rosdep/sources.cache/b67f64890f6d5269c51a68744711f115a8f207ae.pickle
loading cached data source:
	https://raw.githubusercontent.com/ros/rosdistro/master/melodic/distribution.yaml
	/root/.ros/rosdep/sources.cache/92e5d8c5f0d9455fe3a14e3dd1f7b90ca9ae78a8.pickle
```

Is there a way to change the URL from master to a commit SHA or tag instead of master? That way I could download old yaml files that still work fine in melodic and bionic.

mjcarroll, 2024-03-28:
> do you think we should restore these keys for end-of-life distributions?

I don't think that it makes particular sense to keep maintaining these if they are clearly end of life.  Users should be able to use previous tags to preserve their current status, but consider migrating soon.

wico-silva, 2024-04-06:
@mjcarroll, that's reasonable. How can users do this though?

> Users should be able to use previous tags to preserve their current status

mjcarroll, 2024-04-09:
Oh, perhaps I was mistaken here.  I thought that rosdep would let you target previous tags.  @cottsay is this something in a PR that I am mis-remembering?

mjcarroll, 2024-04-09:
Well, with that PR landed, I think that resolves this.  @wico-silva and @Louis-PR feel free to try that out and open issues against the documentation if you continue to have issues.

closed 2024-04-09 by mjcarroll