ros/rosdistro #41622: Octomap as released into Jazzy and Rolling is ABI-incompatible with system versions

clalancette, 2024-06-14:
It turns out that in Ubuntu 24.04, the version of octomap released into the operating system is 1.9.7.  However, we also vendor octomap in both [Jazzy](https://github.com/ros/rosdistro/blob/b68367a34cfaed8c91a8dcd2c440e13710346a81/jazzy/distribution.yaml#L4070-L4088) and [Rolling](https://github.com/ros/rosdistro/blob/b68367a34cfaed8c91a8dcd2c440e13710346a81/rolling/distribution.yaml#L4019-L4037), and that version is version 1.10.0.  This has ended up leading to hacks like in https://github.com/moveit/geometric_shapes/commit/9e862f7b1a4d4930d69036589726ac9058451b4e .

What we should do here instead is to always rely on the operating system version, and not vendor it ourselves (across all ROS distributions).  That will allow us to link with other things from the system without ABI problems.  To do that, we should do the following:

1.  Get octomap released on RHEL-9.  This is our one supported rosdep operating system where it is not available.
1.  Add a liboctomap-dev key into rosdep here in rosdistro.
1.  Update the packages that depend on this to use that new key.  As of this writing, in Rolling those packages are:
    1. hpp-fcl
    1. octomap_ros
    1. gz_dartsim_vendor
    1. mrpt2
    1. moveit_ros_occupancy_map_monitor
    1. moveit_core
    1. dynamic_edt_3d
    1. geometric_shapes
    1. octomap_rviz_plugins
    1. octomap_server
1. Remove the octomap package from rolling, jazzy, iron, and humble.

@rhaschke @henningkayser @wxmerkt FYI.

clalancette, 2024-06-14:
It's been packaged for RHEL-9 now, we just need to wait for it to make it to stable: https://bodhi.fedoraproject.org/updates/FEDORA-EPEL-2024-71ae26cb75

closed 2024-08-26 by clalancette

traversaro, 2024-12-18:
Hello @clalancette, I reached this issue and I was wondering the following: the issue was solved for `rolling` in https://github.com/ros/rosdistro/pull/42561, but it is still present (and it will probably remain present) in `jazzy`, is that correct?

clalancette, 2024-12-18:
> Hello @clalancette, I reached this issue and I was wondering the following: the issue was solved for `rolling` in #42561, but it is still present (and it will probably remain present) in `jazzy`, is that correct?

It is probably possible to fix it in Jazzy as well, though it will require all of the same steps again.