ros/rosdistro #42698: invalid key for pybind11 on RHEL8

christianrauch, 2024-09-02:
With https://github.com/ros/rosdistro/pull/31817 (@cottsay) the `pybind11-dev` key resolves to package `pybind11-devel` on any RHEL except 7. But that package does not exist for RHEL8: https://dl.fedoraproject.org/pub/epel/8/Everything/x86_64/Packages/p/.

cottsay, 2024-09-02:
It was available at the time, but you're right, it's not available right now for RHEL 8's default Python 3 interpreter.

Relevant: https://bugzilla.redhat.com/show_bug.cgi?id=2151596

Have you considered using the `pybind11_vendor` package from libcamera instead of a direct system package dependency?

christianrauch, 2024-09-03:
> Have you considered using the `pybind11_vendor` package from libcamera instead of a direct system package dependency?

libcamera does not come with its own copy or reference of pybind11. Do you mean the [`pybind11_vendor` package](https://github.com/ros2/pybind11_vendor)?

In any case, packages using that rodep key would fail to install the resolved RPM package on RHEL8. I think it would be better to remove that key from RHEL8.

closed 2024-09-06 by clalancette