ros/rosdistro #43395: [Ubuntu Noble] catkin-lint version 1.6.22 is broken and hangs forever

mikaelarguedas, 2024-11-08:
The version of catkin-lint (1.6.22) provided in Ubuntu Noble fails to find packages and hangs forever.
The more recent versions 1.6.23+ do work. (I suppose [this commit](https://github.com/fkie/catkin_lint/commit/4351b1bc753f44fab9fdb812e099199c67b3a911) is needed)

I'm asking here as in the past `catkin_lint` was provided in the ROS apt repository.

Would it be possible to do one of the two following ?
- provide a more recent version of catkin_lint in the ubuntu apt repositories (1.6.25 is in Debian testing) @artivis @florcabral
- provide a more recent version of catkin_lint in the ROS apt repositories @nuclearsandwich @cottsay @tfoote 

Related to https://github.com/ros/rosdistro/pull/41004

Thanks ! :pray:

artivis, 2024-11-08:
Hi @mikaelarguedas, thanks for the ping :+1: .  
Tracking this at [LP#2087555](https://launchpad.net/bugs/2087555) as well. Note that the version is also in Oracular.

artivis, 2024-11-08:
@mikaelarguedas would you happen to have a minimal reproducible example? I gave this a quick try and it seems to work fine.

mikaelarguedas, 2024-11-08:
While trying to provide a simple reproducible example I faced another issue :sweat_smile: 

Simply installing the package as is on a base ubuntu 24.04 image:
```docker
FROM ubuntu:noble

RUN apt update && apt install -y git catkin-lint

WORKDIR /test_ws/src
RUN git clone https://github.com/ros/std_msgs

RUN catkin_lint .
```
results in:
```
 > [5/5] RUN catkin_lint .:                                                                                                                                                             
0.236 catkin_lint: cannot load rosdep database: No module named 'rosdep2'                                                                                                               
0.236 catkin_lint: unknown dependencies will be ignored
0.246 catkin_lint: cannot import 'all': No module named 'distutils'
```
And I can't seem to find a rosdep candidate in the ubuntu packages for noble:
https://packages.ubuntu.com/search?keywords=rosdep&searchon=names&suite=all&section=all

---

I haven't reproduced the freeze in a container yet but have a couple machines with fairly fresh ubuntu installs that do exhibit it, I'll dig a bit more into it in the next couple days

artivis, 2024-11-08:
You may have missed the following: [Changes to ROS Debian Science Support packages](https://discourse.ubuntu.com/t/changes-to-ros-debian-science-support-packages/44497) which explains why rosdep isn't in Noble.

As for the issue with catkin_lint, I can install it in a fresh LXD container and somewhat successfully execute it:

```bash
lxc launch ubuntu:24.04 catkin-lint
lxc shell catkin-lint

ubuntu@catkin-lint:~$ apt update && apt install catkin-lint
...
ubuntu@catkin-lint:~$ mkdir workspace && cd workspace
ubuntu@catkin-lint:~$ git clone https://github.com/ros/geometry2 -b noetic-devel
ubuntu@catkin-lint:~$ catkin_lint .
catkin_lint: cannot load rosdep database: No module named 'rosdep2'
catkin_lint: unknown dependencies will be ignored
test_tf2: CMakeLists.txt(19): error: catkin_add_gtest() used without if(CATKIN_ENABLE_TESTING)
test_tf2: CMakeLists.txt(22): error: catkin_add_gtest() used without if(CATKIN_ENABLE_TESTING)
...
```

mikaelarguedas, 2024-11-08:
Thanks for looking into it!

> You may have missed the following: [Changes to ROS Debian Science Support packages](https://discourse.ubuntu.com/t/changes-to-ros-debian-science-support-packages/44497) which may explain why rosdep isn't in Noble.

oh yeah I did miss that, good news!
Interesting that `catkin-lint` made the cut without rosdep.

----
Reproducing the exact same set of steps on the official ubuntu:noble image:
```
docker run -it --rm ubuntu:noble
# apt update && apt upgrade -y && apt install -y git catkin-lint
# mkdir workspace && cd workspace/
# git clone https://github.com/ros/geometry2 -b noetic-devel
# catkin_lint .

catkin_lint: cannot load rosdep database: No module named 'rosdep2'
catkin_lint: unknown dependencies will be ignored
catkin_lint: cannot import 'all': No module named 'distutils'
```
Note sure what is different between the lxc base image and the docker one.
That specific issue seems to be addressed in https://github.com/fkie/catkin_lint/commit/da9f311d4e9253b11dbb460170d6a8a12b433058 (commit part of 1.6.23 but not 1.6.22 of catkin_lint)

---
Looking into the hanging issue, it may be some conflicting deps on some of our development environment a may not be an issue with the upstream pacage itself :pray: :confused: 
If that is confirmed I will close this,