ros/rosdistro #43795: Add "pdal" to rosdistro/base

jlblancoc, 2024-12-04:
[PDAL](https://pdal.io/) is a C++ library for translating and manipulating [point cloud data](http://en.wikipedia.org/wiki/Point_cloud).

It has potential interest for some ROS packages related to mapping / pointcloud processing, including readers/writers to .las/.laz files.

It's already packaged for Debian / Ubuntu. I have to check whether it's packaged for other distros and then will try to make the PR myself. 
Just wanted to left this here in case someone can take a look and create the PR before that.

clalancette, 2025-01-02:
Sounds good, thanks.  Please feel free to open a rosdep PR when you are ready.

closed 2025-01-02 by clalancette

jlblancoc, 2025-01-02:
@clalancette It seems the package was removed from Debian 2 years ago, despite it's been maintained in Ubuntu (!), so I guess it's not allowed to add a `debian: null` entry, right? :-(