ros/rosdistro #43891: glslc Entry for Ubuntu Jammy

meetgandhi-dev, 2024-12-12:
The **glslc** key was added as a part of PR #40284 . 
The **Ubuntu Jammy** field is currently marked as null, but it is available through the [**vulkan-sdk**](https://community.khronos.org/t/ubuntu-and-glslc/110198/5). Do you think we can add an entry for it to resolve this issue?

@azeey I’m tagging you here for reference, in case you have any insights or suggestions regarding this.

azeey, 2024-12-12:
@meetgandhi-dev There is no package called vulkan-sdk that's shipped with upstream Ubuntu Jammy ([search](https://packages.ubuntu.com/search?keywords=vulkan-sdk&searchon=names&suite=jammy&section=all)), so I'm not sure how we can resolve this.

I also searched for packages that contain `glslc` (the binary file) in Jammy the [results](https://packages.ubuntu.com/search?searchon=contents&keywords=glslc&mode=filename&suite=jammy&arch=any) indicate that there's no such package.

meetgandhi-dev, 2024-12-15:
It is coming from a third-party vendor.
I followed below steps to install the same:
```bash
wget -qO- https://packages.lunarg.com/lunarg-signing-key-pub.asc | sudo tee /etc/apt/trusted.gpg.d/lunarg.asc
sudo wget -qO /etc/apt/sources.list.d/lunarg-vulkan-jammy.list https://packages.lunarg.com/vulkan/lunarg-vulkan-jammy.list
sudo apt update
```

Here is the output dump:
```bash
$ sudo apt-get install shaderc
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  shaderc
0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.
Need to get 6,308 kB of archives.
After this operation, 33.3 MB of additional disk space will be used.
Get:1 https://packages.lunarg.com/vulkan jammy/main amd64 shaderc amd64 2024.3~rc1-1lunarg22.04-1 [6,308 kB]
Fetched 6,308 kB in 5s (1,180 kB/s)  
Selecting previously unselected package shaderc.
(Reading database ... 243932 files and directories currently installed.)
Preparing to unpack .../shaderc_2024.3~rc1-1lunarg22.04-1_amd64.deb ...
Unpacking shaderc (2024.3~rc1-1lunarg22.04-1) ...
Setting up shaderc (2024.3~rc1-1lunarg22.04-1) ...
Processing triggers for libc-bin (2.35-0ubuntu3.8) ...
```

```bash
$ dpkg -S $(which glslc)
shaderc: /usr/bin/glslc
```

meetgandhi-dev, 2024-12-15:
Another point to note is that the repository does not support the arm64 architecture, so I'm unsure whether we can add the entry here.

**N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://packages.lunarg.com/vulkan jammy InRelease' doesn't support architecture 'arm64'**

azeey, 2024-12-17:
I believe, as a policy, we can't use third-party vendors for rosdep, so I think we have to leave it unset for Jammy.

meetgandhi-dev, 2024-12-17:
Okay. Thank you for the clarification.

closed 2024-12-17 by meetgandhi-dev