ros/rosdistro #44272: Reminder to update GPG public keys in time

kgoger, 2025-01-24:
As I could not find an open ticket to handle the upcoming expiration of the GPG keys (2025-06-01) I created one so that we are not running in the same issues as last time https://github.com/ros/rosdistro/pull/29716

```
pub   rsa4096 2019-05-30 [SC] [expires: 2025-06-01]
      C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
uid           [ unknown] Open Robotics <info@osrfoundation.org>

```

Hugal31, 2025-03-18:
When will this be done? And how? By making everyone manually download the new key like last time? Could we instead have a package to automatically update the keys, like for example the `ubuntu-keyring` package?

Anyway, we shouldn't wait for the last minute to renew the key: we need a transition period long enough for everyone to update it so it doesn't cause interruption. For example our robots are offline and connected to the internet for update about once a month, so any transition window smaller than that would require a manual intervention.

HansRobo, 2025-04-11:
I found that ROS PMC has remember and  discussed abut this.
https://docs.google.com/document/d/1TGXYp33N00ZJtilsRzfClFXcSyMa4QKVNmA03Lx9D7w/edit?usp=sharing

According to the minutes, as mentioned by @Hugal31 , a new package called `ros-archive-keyring` will be prepared to handle key updates.

claraberendsen, 2025-05-27:
This should be closed by https://github.com/ros/rosdistro/pull/46048.

facontidavide, 2025-06-01:
soooooooooooooooooooooooooooooooo...

[image]

mikeferguson, 2025-06-01:
https://discourse.ros.org/t/ros-signing-key-migration-guide/43937

facontidavide, 2025-06-02:
thanks a lot, I donn't look that often at ROS Discourse and I missed that announcement.

But I am confused. I am seeing these errors in Jenkins (https://build.ros2.org/) and Github Actions, where I don't have much control.

For the latter, it was reported already, apparently: https://github.com/ros-tooling/setup-ros/issues/815

claraberendsen, 2025-06-05:
@facontidavide the build.ros2.org situation should be resolved as well as the docker images.

clalancette, 2026-02-10:
I believe that this was all fixed by the updated package, so closing this out.  Feel free to reopen if that is not the case.