ros/rosdistro #44471: Using python-cantools-pip as a dependency creates uninstallable deb package

superdyoll, 2025-02-13:
python-cantools-pip is a pip package, and has no equivalent apt package to install (it installs https://github.com/cantools/cantools )

Adding a dependency to python-cantools-pip in a package.xml as an exec_depend (The recommended dependency for python packages. See https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Rosdep.html#exec-depend ) will work, and can be built fine, and will create a deb file. However, the resultant deb file includes a dependency on cantools (which as noted isn't a package that exists in apt) and as such you're left with a deb file that can never be installed.

package.xml
```xml
<exec_depend>python-cantools-pip</exec_depend>
```

Building using
```bash
bloom-generate rosdebian --ros-distro jazzy
fakeroot debian/rules binary
```

Interrogating resultant deb file
```bash
$ dpkg -I ros-jazzy-test_0.0.0-0noble_amd64.deb
...
Depends: cantools
...
```

If you try to install said package you get
```bash
The following packages have unmet dependencies.
 ros-jazzy-test : Depends: cantools but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Perhaps we're misusing this, but we've not had any other issues with other packages, and we're not sure where it's getting the cantools dep from as rosdep definitely has it listed as coming from pip (See https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml#L869C1-L869C20 )

If this isn't the correct place to raise this issue, please advise where the correct location is? Thanks

tonybaltovski, 2025-03-14:
I ran into this myself.  AFIK apt can't install pip packages.  The default path forward for me is to make a debian of the pip package.  I started a issue here: https://github.com/cantools/cantools/issues/224.  I think the person who commented also look at the same issue.

superdyoll, 2025-03-16:
@tonybaltovski in my mind the problem here is two fold, firstly the packages shouldn't be including pip packages as an apt dependency if rosdistro knows that it's a pip dependency (which this case of an example of as it's specifically in the yaml as coming from pip). Fixing that would at least make the package installable, although one would need to be careful to install any sub-dependencies with pip manually after installing an apt package that uses pip packages.

To fix the second problem, ros packages that have pip dependencies should probably automatically have a dependency on the apt pip/python deb package, and then auto-generate a post install (postinst) script that installs all the pip packages that are depended on, so that when you install a ros package with pip dependencies they automatically get installed when you install the overall apt package. (I think it'd need some careful thinking about, but I think you'd just have to make an assumption to install them as a pip system package rather than thinking about venv's etc)

It probably is good to encourage 3rd party packages like cantools to create debian packages because ubuntu is such a popular distro, but given they're external I don't think rosdistro can easily enforce them to support a certain package manager. Given python is one of the official ros2 languages, and pip is the default package manager for python, there probably aught to be better support for including pip only packages in the packaging of ros2 packages (and it definitely shouldn't break the packages if you use the package.xml in the recommended way that works with rosdep 😛)

mjcarroll, 2025-05-06:
Packages built by the buildfarm explictly _cannot_ depend on pip-only keys. I suppose there is a case where a pure python package doesn't have tests that would cause it to fail the build of the package, so the package builds/installs successfully.

When merging the pip rosdistro keys, we include this standard disclaimer to cover this scenario.

```
#Standard pip disclaimer: ROS packages that depend on pip keys cannot be released into a ROS distribution. They can only be depended on by from-source builds. Because of this, system packages are highly preferred to pip packages.
```

If you want a package that can successfully be built and deployed as a debian package, you must ensure that all of the dependencies are debian packages, either from upstream or as a vendor package.

> Given python is one of the official ros2 languages, and pip is the default package manager for python, there probably aught to be better support for including pip only packages in the packaging of ros2 packages

This is fundamentally a limitation of the distribution model, rather than specifically a ROS issue. For Debian and RHEL packages, we are following the packaging guidelines for those distributions.  For installing pip packages, this is supported via the `rosdep` tool, but won't be possible to support via debian or rpm packages.

closed 2025-05-06 by mjcarroll