ros/rosdistro #44826: Unable to find libqt5core5a on ArmHF version of Ubuntu 22.04

yuqingli05, 2025-03-19:
Unable to find libqt5core5a on ArmHF version of Ubuntu 22.04

sloretz, 2025-03-19:
Are you saying you tried using rosdep to install the `libqt5-core` rosdep key on Ubuntu 24.04 (Noble), and it couldn't `apt-get install` the package named `libqt5core5a`? 

https://github.com/ros/rosdistro/blob/2132ffa32b587668143e3a566a7767c2bce32728/rosdep/base.yaml#L5878-L5890

It looks like the key is available for Ubuntu 22.04, but not Ubuntu 24.04

* Ubuntu 22.04 https://packages.ubuntu.com/jammy/libqt5core5a

It seems like the equivalent package in Noble might be `libqt5core5t64`. https://packages.ubuntu.com/noble/libqt5core5t64 . 

Please open a PR to use `libqt5core5t64` in Ubuntu Noble, and `libqt5core5a` in previous ubuntu distros.

Example: https://github.com/ros/rosdistro/blob/2132ffa32b587668143e3a566a7767c2bce32728/rosdep/base.yaml#L2784-L2788

yuqingli05, 2025-03-20:
> Are you saying you tried using rosdep to install the `libqt5-core` rosdep key on Ubuntu 24.04 (Noble), and it couldn't `apt-get install` the package named `libqt5core5a`?
> 
> [rosdistro/rosdep/base.yaml](https://github.com/ros/rosdistro/blob/2132ffa32b587668143e3a566a7767c2bce32728/rosdep/base.yaml#L5878-L5890)
> 
> Lines 5878 to 5890 in [2132ffa](/ros/rosdistro/commit/2132ffa32b587668143e3a566a7767c2bce32728)
> 
>  libqt5-core: 
>    alpine: [qt5-qtbase] 
>    arch: [qt5-base] 
>    debian: [libqt5core5a] 
>    fedora: [qt5-qtbase] 
>    freebsd: [qt5-core] 
>    gentoo: ['dev-qt/qtcore:5'] 
>    nixos: [qt5.qtbase] 
>    openembedded: [qtbase@meta-qt5] 
>    opensuse: [libQt5Core5] 
>    rhel: [qt5-qtbase] 
>    slackware: [qt5] 
>    ubuntu: [libqt5core5a] 
> It looks like the key is available for Ubuntu 22.04, but not Ubuntu 24.04
> 
> * Ubuntu 22.04 https://packages.ubuntu.com/jammy/libqt5core5a
> 
> It seems like the equivalent package in Noble might be `libqt5core5t64`. https://packages.ubuntu.com/noble/libqt5core5t64 .
> 
> Please open a PR to use `libqt5core5t64` in Ubuntu Noble, and `libqt5core5a` in previous ubuntu distros.
> 
> Example:
> 
> [rosdistro/rosdep/base.yaml](https://github.com/ros/rosdistro/blob/2132ffa32b587668143e3a566a7767c2bce32728/rosdep/base.yaml#L2784-L2788)
> 
> Lines 2784 to 2788 in [2132ffa](/ros/rosdistro/commit/2132ffa32b587668143e3a566a7767c2bce32728)
> 
>  ubuntu: 
>    bionic: [libboost-atomic1.65.1] 
>    focal: [libboost-atomic1.71.0] 
>    jammy: [libboost-atomic1.74.0] 
>    noble: [libboost-atomic1.83.0]

Thank you, it has been successful. Additionally, libqt5gui5 should also be replaced with libqt5gui5t64.

omarzayedbeshir, 2025-04-08:
Addressed here: https://github.com/ros/rosdistro/pull/45035

pjreed, 2025-05-13:
This update breaks rosdep on Ubuntu 18.04 / ROS Melodic because it needs `libqt5core5a`, not `libqt5core5t64`.  I know it's old and no longer supported, but we still have systems using it and this breaks the build process on them.

sloretz, 2025-05-13:
> This update breaks rosdep on Ubuntu 18.04 / ROS Melodic because it needs libqt5core5a, not libqt5core5t64. I know it's old and no longer supported, but we still have systems using it and this breaks the build process on them.

Based on #18296, I'd recommend making your rosdep sources use the [last ROS Melodic tag](https://github.com/ros/rosdistro/tree/melodic/2023-06-27) instead of `master`. https://docs.ros.org/en/independent/api/rosdep/html/sources_list.html

Closing this issue as completed by #45035

closed 2025-05-13 by sloretz