ros/rosdistro #46092: binary build job for py_binding_tools fails

rhaschke, 2025-05-26:
The binary build jobs for py_binding_tools (rolling) fail on the ROS build farm due to misconfigured dependencies in the .dsc file:
[`KeyError: "No packages available for 'ros-one-ament-cmake'"`](https://build.ros2.org/job/Rbin_uN64__py_binding_tools__ubuntu_noble_amd64__binary/51/console)

The reason was a misconfiguration of rosdep on my machine when I triggered bloom-release. I fixed that as well as the release commit.

Could anybody please retrigger the source build:
https://build.ros2.org/view/Rsrc_uN/job/Rsrc_uN__py_binding_tools__ubuntu_noble__source

tfoote, 2025-05-29:
Patching in place is not a reliable way to make sure that the build comes out consistently. Please rerun bloom and it will generate a new version with a higher build increment (the number that is usually 0 at the end of the version) so that the buildfarm will detect the change and process it throughout the pipeline. There's a lot of caching and without the increment you can get interleaved content.

rhaschke, 2025-05-30:
Trying to run bloom again, I get the following error:
```
####
#### Generating 'noble' debian for package 'py_binding_tools' at version '2.0.2-2'
####
 [git-bloom-patch export]: 'execute_command' failed to call 'git format-patch -M -B 429aa16...debian/rolling/noble/py_binding_tools' which had a return code (128):
 [git-bloom-patch export]: ```
fatal: ambiguous argument '429aa16...debian/rolling/noble/py_binding_tools': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'

 [git-bloom-patch export]: ```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/bloom/commands/git/generate.py", line 101, in try_execute
    retcode = func(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/bloom/logging.py", line 206, in decorated
    return f(*args, **kwds)
  File "/usr/lib/python3/dist-packages/bloom/commands/git/patch/export_cmd.py", line 53, in export_patches
    execute_command(cmd, cwd=directory)
  File "/usr/lib/python3/dist-packages/bloom/util.py", line 451, in execute_command
    raise CalledProcessError(result, cmd)
subprocess.CalledProcessError: Command 'git format-patch -M -B 429aa16...debian/rolling/noble/py_binding_tools' returned non-zero exit status 128.

Error calling git-bloom-patch export: Command 'git format-patch -M -B 429aa16...debian/rolling/noble/py_binding_tools' returned non-zero exit status 128.
git-bloom-patch export returned exit code (128)
❌  Error running command '['/usr/bin/git-bloom-generate', '-y', 'rosdebian', '--prefix', 'release/rolling', 'rolling', '-i', '2', '--os-name', 'ubuntu']'
Release failed, exiting.
```

tfoote, 2025-07-14:
Have you resovled this? Did you make a new release with a higher version?

If you reran bloom with out increasing the version number it may be that the base tarball is the old version and it's trying to overlay the diff based on the new tarball, but if they don't match the source tarball then it will collide and fail to apply.

rhaschke, 2025-08-12:
Looks like I screwed the release repo. bloom-release failed even with a new release number. Finally fixed now.

closed 2025-08-12 by rhaschke