ros/rosdistro #46433: `bloom-release` cannot find `rcl_logging_syslog` that is already indexed.

fujitatomoya, 2025-06-10:
related to https://github.com/ros/rosdistro/pull/46239

i was trying to `bloom-release` along with https://docs.ros.org/en/rolling/How-To-Guides/Releasing/First-Time-Release.html#bloom-release

```console
tomoyafujita@~/DVT/03_OSS/github.com/fujitatomoya/rcl_logging_syslog >bloom-release --new-track --rosdistro rolling --track rolling rcl_logging_syslog
ROS Distro index file associate with commit '83cc9c1a7d9e30479c7017068510f332fedc157e'
New ROS Distro index url: 'https://raw.githubusercontent.com/ros/rosdistro/83cc9c1a7d9e30479c7017068510f332fedc157e/index-v4.yaml'
Specified repository 'rcl_logging_syslog' is not in the distribution file located at 'https://raw.githubusercontent.com/ros/rosdistro/83cc9c1a7d9e30479c7017068510f332fedc157e/rolling/distribution.yaml'
Did you mean one of these: 'rcl_logging_syslog', 'rcl_logging', 'rcl_logging_rcutils'?
Could not determine release repository url for repository 'rcl_logging_syslog' of distro 'rolling'
You can continue the release process by manually specifying the location of the RELEASE repository.
To be clear this is the url of the RELEASE repository not the upstream repository.
For release repositories on GitHub, you should provide the `https://` url which should end in `.git`.
Here is the url for a typical release repository on GitHub: https://github.com/ros-gbp/rviz-release.git
==> Looking for a release of this repository in a different distribution...
No reasonable default release repository url could be determined from previous releases.
Release repository url [press enter to abort]:
No release repository url given, aborting.
```

the error message is strange because https://raw.githubusercontent.com/ros/rosdistro/83cc9c1a7d9e30479c7017068510f332fedc157e/rolling/distribution.yaml does have `rcl_logging_syslog`.

did i mess up something here?

Note: https://docs.ros.org/en/rolling/How-To-Guides/Releasing/First-Time-Release.html#bump-the-package-version requires uncheck `Require a pull request before merging` for the repository rule, otherwise the remote rejects the change. this would be good to add the documentation.

christophebedard, 2025-06-10:
I think you need to create a release repo first, see: https://docs.ros.org/en/rolling/How-To-Guides/Releasing/Release-Team-Repository.html#create-a-new-release-repository

fujitatomoya, 2025-06-10:
@christophebedard thanks for the information, i created https://github.com/ros2-gbp/ros2-gbp-github-org/issues/792 that should fix this failure after closed.

closed 2025-06-10 by fujitatomoya