ros/rosdistro #46814: ROS2 Humble Release Job failing for `adi_iio`

Adrian-Stanea, 2025-07-08:
A release PR for the iio_ros2 package into the humble distribution was recently merged: https://github.com/ros/rosdistro/pull/46666 . However, it appears that the subsequent Jenkins CI release job is consistently failing during the `install` phase with a `CMake Error` indicating that a config directory cannot be found.

The specific error messages from the CI console output are:
```bash
Make Error at cmake_install.cmake:619 (file):
09:40:59   file INSTALL cannot find "/tmp/binarydeb/ros-humble-adi-iio-1.0.0/config":
09:40:59   No such file or directory.
09:40:59 
09:40:59 
09:40:59 make[2]: Leaving directory '/tmp/binarydeb/ros-humble-adi-iio-1.0.0/.obj-x86_64-linux-gnu'
09:40:59 make[2]: *** [Makefile:113: install] Error 1
09:40:59 dh_auto_install: error: cd .obj-x86_64-linux-gnu && make -j1 install DESTDIR=/tmp/binarydeb/ros-humble-adi-iio-1.0.0/debian/ros-humble-adi-iio AM_UPDATE_INFO_DIR=no returned exit code 2
09:40:59 make[1]: Leaving directory '/tmp/binarydeb/ros-humble-adi-iio-1.0.0'
09:40:59 make[1]: *** [debian/rules:66: override_dh_auto_install] Error 25
09:40:59 make: *** [debian/rules:27: binary] Error 2
09:40:59 dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2
09:40:59 Traceback (most recent call last):
09:40:59   File "/tmp/ros_buildfarm/ros_buildfarm/binarydeb_job.py", line 149, in build_binarydeb
09:40:59     subprocess.check_call(cmd, cwd=source_dir, env=env)
09:40:59   File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
09:40:59     raise CalledProcessError(retcode, cmd)
09:40:59 subprocess.CalledProcessError: Command '['dpkg-buildpackage', '-b', '-us', '-uc']' returned non-zero exit status 2.
09:40:59 # END SUBSECTION
09:40:59 
09:40:59 --------------------------------------------------------------------------------------------------
09:40:59 `dpkg-buildpackage -b -us -uc` failed.
09:40:59 This is usually because of an error building the package.
```

## Context
Prior to initiating the release, I followed the standard local packaging procedure as described in: https://docs.ros.org/en/humble/How-To-Guides/Building-a-Custom-Deb-Package.html .
As a result I obtained `.deb` pacakges for that were installed locally into a clean Docker environment.
I can confirm that the package installed succesfully, including the `config` folder which the CI is complaining about. This is shown below:
``` bash
adi_iio ll 
total 48K
drwxr-xr-x 2 root root 4.0K Jul  8 11:35 cmake
drwxr-xr-x 2 root root 4.0K Jul  8 11:35 config
drwxr-xr-x 2 root root 4.0K Jul  8 11:35 environment
drwxr-xr-x 2 root root 4.0K Jul  8 11:35 launch
-rw-r--r-- 1 root root 1.6K Jun 27 10:24 local_setup.bash
-rw-r--r-- 1 root root  191 Jun 27 10:24 local_setup.dsv
-rw-r--r-- 1 root root 5.1K Jun 27 10:24 local_setup.sh
-rw-r--r-- 1 root root 2.0K Jun 27 10:24 local_setup.zsh
-rw-r--r-- 1 root root  148 Jun 27 10:24 package.dsv
-rw-r--r-- 1 root root 2.0K Jun 27 10:24 package.xml
drwxr-xr-x 2 root root 4.0K Jul  8 11:35 srv
➜  adi_iio ls config 
adi_iio.yaml
➜  adi_iio pwd                  
/opt/ros/humble/share/adi_iio
➜  adi_iio 
```

We also attempted to reproduce the release job locally using the ros_buildfarm documentation (https://github.com/ros-infrastructure/ros_buildfarm/blob/master/doc/jobs/release_jobs.rst). However, the process described seems to be outdated for newer distributions like Humble, as we were only able to get it working for the Noetic distribution.

Given that the locally generated .deb installs correctly and contains the necessary config directory, it is unclear why the build farm environment is failing to find this directory during its installation process.

NOTE: the package defines a set of tests that are failing within the CI job since they require HW to excute. However the worflow continues even after the tests fail, so I'm not sure if this is an issue that needs to be addressed (we can use cmake flags that are disabled by default to avoid executing the tests that require HW). However I'm not sure if this is the actual problem.

Any assistance in diagnosing and resolving this issue would be greatly appreciated.

christophebedard, 2025-07-11:
Can you try changing your `install(...)` from:

```cmake
install(DIRECTORY config/
  DESTINATION share/${PROJECT_NAME}/config
)
```

to this?

```cmake
install(DIRECTORY config
  DESTINATION share/${PROJECT_NAME}
)
```

See this one: https://github.com/ros2/rmw_zenoh/blob/1f1dafdb71ae63e0b6b4db38a02fefc7df195397/rmw_zenoh_cpp/CMakeLists.txt#L133-L136

Adrian-Stanea, 2025-07-14:
I am not sure that will help since for release `1.0.0` i had the install rule [like this](https://github.com/ros2-gbp/adi_iio-release/blob/051a69392a8391725cd51a60c7ffe515c0cb6177/CMakeLists.txt#L74):
```cmake
install(DIRECTORY
  launch
  config
  DESTINATION share/${PROJECT_NAME}/
)
``` 
The problem would still persist since the `config` directory is not being copied to the release repository by `blooom`. 
Are there any guidelines for doing manual changes to the release repository?

christophebedard, 2025-07-16:
> I am not sure that will help since for release 1.0.0 i had the install rule [like this](https://github.com/ros2-gbp/adi_iio-release/blob/051a69392a8391725cd51a60c7ffe515c0cb6177/CMakeLists.txt#L74):

but there's a trailing slash there, so it is not the same thing

> Are there any guidelines for doing manual changes to the release repository?

I don't think that's a good idea.

Adrian-Stanea, 2025-07-29:
Update: after going through the source code of [bloom](https://github.com/ros-infrastructure/bloom?tab=readme-ov-file), I managed to find the issue: **my machine had a global `.gitignore` that matched the single file contained within the `config` directory of the project.** 

I noticed that the `bloom-export-upstream` step properly generated the archive containing the directory, however at [this step of `git-bloom-import-upstream`](https://github.com/ros-infrastructure/bloom/blob/6199b2a76f50f120a398c48c44b1e1ffd6c5f38f/bloom/commands/git/import_upstream.py#L151) (using `--force` might make sense since we are only staging items from the archive) , the global `.gitignore` that matched my file excluded the `config` directory entirely and hence could not be tracked by the git add command.

closed 2025-07-29 by Adrian-Stanea