ros/rosdistro #47101: Missing boost package keys on Debian Bookworm

johnwason, 2025-07-28:
I received this error using bloom release:

```
Could not resolve rosdep key 'libboost-system' for distro 'bookworm':
No definition of [libboost-system] for OS version [bookworm]
	rosdep key : libboost-system
	OS name    : debian
	OS version : bookworm
	Data:
arch:
		- boost-libs
		debian:
		  bullseye:
		  - libboost-system1.74.0
		  buster:
		  - libboost-system1.67.0
		fedora:
		- boost-system
		gentoo:
		- dev-libs/boost
		nixos:
		- boost
		openembedded:
		- boost@openembedded-core
		opensuse:
		- libboost_system1_66_0
		rhel:
		- boost-system
		ubuntu:
		  bionic:
		  - libboost-system1.65.1
		  focal:
		  - libboost-system1.71.0
		  jammy:
		  - libboost-system1.74.0
		  noble:
		  - libboost-system1.83.0
		
Failed to resolve libboost-system on debian:bookworm with: Error running generator: Failed to resolve rosdep key 'libboost-system', aborting.
libboost-system is depended on by these packages: ['robotraconteur']
<== Failed
Some of the dependencies for packages in this repository could not be resolved by rosdep.
❌  The following generator action reported that it is missing one or more

```

It appears the boost package keys are missing for Debian `bookworm`, but they are available for `buster` and `bullseye`.

christophebedard, 2025-07-30:
Looks like it's just missing `bookworm: [libboost-system1.74.0]`: https://pkgs.org/search/?q=libboost-system

Would you be willing to submit a PR?

johnwason, 2025-07-30:
There are a lot of similar keys missing for the other boost libraries - libboost-system is just one example.

I will take a look at a PR.

closed 2025-10-16 by emersonknapp