ros/rosdistro #47160: Build failed in Jenkins due to missing libraries

nevoded, 2025-07-30:
Building our package (inuitive-camera-ros2) fails.
Build status: https://build.ros2.org/job/Hbin_uJ64__inuros2__ubuntu_jammy_amd64__binary/18/
Our package dependant on two libraries from our SDK:
libInuStreams.so
libInuCommonUtilities.so

These libraries are part of our existing installation of inuitive-camera-ros2:
See on our README.md section, on https://bitbucket.org/inuitive/inuros2.git  
Under "Installation on Ubuntu"->"Step 3: Install Inuitive ROS2 wrapper (InuROS)"

I can add the libraries to the repository but I rather not add binaries to the repository.
Any suggestions on how to tackle this issue?

Does connection to our artifactory using conan (through cmake) is a valid solution on the build farm environment?

christophebedard, 2025-07-30:
Thanks for opening an issue.

I believe you could indeed add something to your package's `CMakeLists.txt` to download the right (compiled) libraries. However, I'm not sure if we generally accept downloading binaries from your company's Artifactory instance (as opposed to github.com). Let me check with others.

One thing to note is that the license of those libraries has to be clearly noted in the `package.xml` file of the package that downloads them. It has to be clear enough so that users can know if they can use/distribute/etc. your proprietary libs.

christophebedard, 2025-07-30:
> I believe you could indeed add something to your package's `CMakeLists.txt` to download the right (compiled) libraries. However, I'm not sure if we generally accept downloading binaries from your company's Artifactory instance (as opposed to github.com). Let me check with others.

Can you share roughly what this would look like in CMake?

mjcarroll, 2025-07-31:
We can't distribute proprietary binaries via the buildfarm without either A) a permissive license attached to them or B) an agreement in place with the OSRF.

nevoded, 2025-08-03:
I'm working on adding the source code of the libraries to our inuros2 repository so they will be built as part of our ros package. Will update soon.

tfoote, 2025-08-05:
It's good for the clarifications above about binary options, but they're all challenging to keep things reproducible especially on potentially very different networking environments and across different platforms. (You need to build just as many binary so as have logic to pull the right one per target arch etc.)

> I'm working to add the source code of the libraries to our inuros2 repository so they will be built as part of our ros package. Will update soon.

This is probably the best solution, it will let us distribute it and make sure it's available on all platforms and architectures. Alternatively you could release the library as a separate package onto the buildfarm and then add a dependency onto it.