ros/rosdistro #47263: OpenEmbedded repository checks disabled due to API errors

cottsay, 2025-08-08:
The rosdep repository checks for OpenEmbedded are failing due to 403 errors returned from the layerindex API. I can't reproduce the problems locally, so I'm disabling the check until we can figure out how to handle this.

robwoolley, 2025-08-12:
I asked about this on the Yocto Technical Call.  The OpenEmbedded Layer Index is behind Cloudflare.  It would have whatever protections are automatically applied to it.  There is no other proof-of-work challenge that we are aware of.

I will follow up with the LF helpdesk to see about how we could possibly add the GHA invocations to the allow list.

cottsay, 2025-08-12:
Thanks. If this is indeed some sort of traffic mitigation, I can only say that I sympathize with the struggle and I'm willing to make some changes to the checker to authenticate, differentiate, or reduce our traffic.